ENSTABretagneRobotics/razor_imu_9dof #23: frame_id is hard coded

nwotero, 2015-07-17:
It would be very helpful and convenient if the frame_id for the imu messages was exposed as a parameter. If a user has more than one imu, they would want to give the messages different frame_ids.

KristofRobot, 2015-07-18:
@nwotero - I think that is an excellent idea! Care to provide a patch?

nwotero, 2015-07-18:
Sure, I'll make a pull request this week once I've verified it works.

Jorge-C, 2015-11-25:
What's the best approach to run several (2) razor IMUs in the same system currently?

I'm thinking about having two copies of the package, and manually editing the code so that I can launch two nodes with different names/published channels, but I wonder what would be a better way to do it. Thanks!

I'd be happy to submit a PR if it requires some improvements in your code.

KristofRobot, 2015-11-25:
@Jorge-C - in principle, you would run multiple instances of the same node, using remapping - see e.g. http://answers.ros.org/question/160071/running-multiple-instances-of-same-node/

Now, I never tried this myself, and most probably the code will need some tuning to support this in an optimal way. Feel free to try it out, and submit a PR!

Jorge-C, 2015-11-25:
Thanks! I'll try it out and report back.

mxochicale, 2016-08-12:
Dear All,

I am very thankful to the developers of this ROS driver. Your well documented README files and source-code is helping me a lot to achieve my scientific endeavours in Robotics. Going back to the purpose of this post. I had the same query as @Jorge-C in order to run more than two razor_imu_9dof sensors. I therefore managed to collect data from two razor_imu_9dof sensors using the following launch file (if anyone is interested, I will happily share the tunning source-code). I am however not following the suggestion of @KristofRobot using remap to run multiple instance of the same node. Honestly, I have very little knowledge of the use of ROS (but I am learning) to which I am not really sure whether
what I am doing might be the better approach to run two or more synchronised sensors. Do you think what I am doing is a good way to collect data from two or more sensors?.

Also, I want to point out that a drift between sensors is presented. To present evidence of this I put two sensors on the table and I hit the table every approximately 30 seconds during 3 minutes seconds. Then, I plot the ACC_Z axes of both sensors 
[image]

I have notice another thing which is that the sample number of both sensors is not the same.

One of my ideas to deal with this technical issue is to use a sort of hand-shake between the razor_imu_9dof and the machine.

I will appreciate if you have any commend or advice in order to solve the problem with
the drift or the syncronisation.

Again, thank you very much!
Miguel

> FILENAME: razor-pub2imus.launch
`
<launch>
<arg name="razor_config0" default="$(find razor_imu_9dof)/config/razor0.yaml"/>
<arg name="razor_config1" default="$(find razor_imu_9dof)/config/razor1.yaml"/>
  <node pkg="razor_imu_9dof" type="imu_node.py" name="imu_node0" ns="ns0" output="screen">
    <rosparam file="$(arg razor_config0)" command="load"/>
  </node>
  <node pkg="razor_imu_9dof" type="imu_node.py" name="imu_node1" ns="ns1" output="screen">
    <rosparam file="$(arg razor_config1)" command="load"/>
  </node>
</launch>
`