ENSTABretagneRobotics/razor_imu_9dof #25: roslaunch razor_imu_9dof razor-pub.launch error

491734045, 2016-02-25:
File "/home/ros-laser/catkin_ws/src/razor_imu_9dof/nodes/imu_node.py", line 230, in <module>
    yaw_deg = -float(words[0])
ValueError: could not convert string to float: #YPR=53.53

the python version I installed is Python 2.7.6, is the python version not proper?

KristofRobot, 2016-02-25:
>>#YPR=53.53

Seems that your razor firmware is outputting data in the wrong format - it should output the "#YPRAG=..." format instead.

Are you sure you are using the custom razor_imu_9dof firmware [1], and NOT the original razor firmware [2], as explained in the README [3]?

[1] https://github.com/KristofRobot/razor_imu_9dof/tree/indigo-devel/src/Razor_AHRS
[2] https://github.com/ptrbrtz/razor-9dof-ahrs
[3] https://github.com/KristofRobot/razor_imu_9dof

KristofRobot, 2016-10-18:
Closing due to inactivity

closed 2016-10-18 by KristofRobot