ENSTABretagneRobotics/razor_imu_9dof #27: roslaunch razor_imu_9dof razor-pub.launch error

carl1218, 2016-11-04:
hi,
My system is 64 bits.i used the  KristofRobot/razor_imu_9dof  firmware.
when i run the launch，I got the following error.
[INFO] [WallTime: 1478277847.034457] Publishing IMU data...
Traceback (most recent call last):
  File "/home/catkin_imu/src/razor_imu_9dof/nodes/imu_node.py", line 228, in <module>
    yaw_deg = -float(words[0])
ValueError: could not convert string to float: 
[imu_node-2] process has died [pid 8670, exit code 1, cmd /home/catkin_imu/src/razor_imu_9dof/nodes/imu_node.py __name:=imu_node __log:=/home/.ros/log/de7abd92-a2ad-11e6-ac74-000c2971e5f1/imu_node-2.log].
log file: /home/.ros/log/de7abd92-a2ad-11e6-ac74-000c2971e5f1/imu_node-2*.log`

KristofRobot, 2016-11-05:
Can you confirm you are loaded the custom firmware (from src/Razor_AHRS/Razor_AHRS.ino) on the IMU? See "Install Arduino firmware" step in the README.

DavidHan008, 2017-07-19:
i have some problems, when i use arduino upload the Razor_AHRS.ino, it has some errors
as followers:

Binary sketch size: 21,538 bytes (of a 30,720 byte maximum)
avrdude: stk500_initialize(): (a) protocol error, expect=0x14, resp=0x23
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x59

rcpilotp51, 2018-02-03:
i get the error

[display_3D_visualization_node-3] process has died [pid 23261, exit code 1, cmd /opt/ros/kinetic/share/razor_imu_9dof/nodes/display_3D_visualization.py __name:=display_3D_visualization_node __log:=/home/kbrown/.ros/log/d7cd8e46-087a-11e8-8f0c-809b20e2e254/display_3D_visualization_node-3.log].
log file: /home/kbrown/.ros/log/d7cd8e46-087a-11e8-8f0c-809b20e2e254/display_3D_visualization_node-3*.log

and cant get anything out ort the serial monitor

CapitalZe, 2018-05-04:
Hello,

I get the following:

ERROR: cannot launch node of type [sparkfun_9dof_razor_imu_m0/driver_node]: sparkfun_9dof_razor_imu_m0
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/nvidia/racecar-ws/src
ROS path [2]=/opt/ros/kinetic/share

I have the appropriate Firmware.

Nothing seems to work. The 3D vizualiser also does not display anything, yet the Arduino IDE does display the imu values.

Any ideas?

Cheers!

jontromanab, 2018-05-08:
@CapitalZe  it seems like the package is not properly installed or the system can not find the path to the package. Try a "roscd razor_imu_9dof" and see if it can find the package

aarontan94, 2018-08-09:
@KristofRobot I am arriving at the same error as @carl1218 and I have already loaded the custom software on the IMU. Was this ever solved? Please help.

djl806943371, 2019-11-22:
> i get the error
>
> [display_3D_visualization_node-3] process has died [pid 23261, exit code 1, cmd /opt/ros/kinetic/share/razor_imu_9dof/nodes/display_3D_visualization.py [...]

I got the same error, have you solved it?