ENSTABretagneRobotics/razor_imu_9dof #28: Support for current boards?

falkoschindler, 2016-12-15:
The [documentation](http://wiki.ros.org/razor_imu_9dof) says:

> Currently the firmware supports the boards with SparkFun product numbers SEN-10125 and SEN-10736.

Unfortunately, [both](https://www.sparkfun.com/products/10125) [products](https://www.sparkfun.com/products/10736) have been retired. Does razor_imu_9dof support newer boards like the current [SEN-14001](https://www.sparkfun.com/products/14001) or even the cheaper [SEN-13944](https://www.sparkfun.com/products/13944) with I2C interface?

KristofRobot, 2016-12-17:
@falkoschindler 

I have not tried one of those newer products, but the hardware seems significantly different. The firmware (in src/Razor_AHRS) will definitely need updating. You might want to raise the question at https://github.com/ptrbrtz/razor-9dof-ahrs - although that repo seems to be inactive, unfortunately.

bpinaya, 2017-06-02:
That is so sad, just upgraded to the SEN-14001, was looking forwards to have it running with ROS but it seems that the changes needed would be quite significant, maybe we can try and ping Sparkfun?

ruffsl, 2017-06-02:
@bpinaya , related? 
https://github.com/Razor-AHRS/razor-9dof-ahrs/issues/57

bpinaya, 2017-06-04:
@ruffsl Thanks! I'll check it out! Btw I am a huge fan of your docker with ros work! Awesome job indeed.

Murphy3220, 2017-09-21:
I have updated the Razor AHRS to work with SEN-14001. All the credit goes to the original coder. You can find the updated code at https://github.com/Murphy3220/Razor-AHRS-SEN-14001 . It only works for the SEN-14001.

lebarsfa, 2017-11-24:
ROS users might also be interested in that version : https://github.com/lebarsfa/razor-9dof-ahrs .

closed 2018-01-20 by KristofRobot

PratibhaT, 2018-04-04:
How Do i mount this sensor: SEN-14001 on my robot?
The gyro and accelerometer axis is different from that of magnetometer axis?

PR #34 (2018-01-20): Added support of SEN-14001 (9DoF Razor IMU M0)
Resolves #28. The firmware modifications to support the M0 were originally made from the latest version of https://github.com/Razor-AHRS/razor-9dof-ahrs, forked in https://github.com/lebarsfa/razor-9dof-ahrs. Then, the ROS-specific modifications from https://github.com/KristofRobot/razor_imu_9dof were added to add ROS compatibility. Finally, following a suggestion on https://answers.ros.org/question/261398/sparkfun-9dof-razor-imu-sen-14001/, this pull request is proposed, and a pull request to https://github.com/Razor-AHRS/razor-9dof-ahrs will be  [... cut]