ENSTABretagneRobotics/razor_imu_9dof #39: Something wrong with the python visual module

amitfishy, 2018-06-01:
Hi

I try running 
```
roslaunch razor_imu_9dof razor-display.launch
```

And the publishing works just fine, I can see the published topic /imu.

The problem is in the visualization. I get the following error:
```
SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    display_3D_visualization_node (razor_imu_9dof/display_3D_visualization.py)

auto-starting new master
process[master]: started with pid [5024]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to f3115532-6588-11e8-ae14-a44cc847fdbe
process[rosout-1]: started with pid [5037]
started core service [/rosout]
process[display_3D_visualization_node-2]: started with pid [5044]
Traceback (most recent call last):
  File "/home/amitsinha/catkin_ws/src/razor_imu_9dof/nodes/display_3D_visualization.py", line 31, in <module>
    from visual import *
  File "/usr/lib/python2.7/dist-packages/visual/__init__.py", line 66, in <module>
    from visual.ui import display
  File "/usr/lib/python2.7/dist-packages/visual/ui.py", line 3, in <module>
    import materials
  File "/usr/lib/python2.7/dist-packages/visual/materials.py", line 152, in <module>
    tx_turb3 = raw_texture( data=reshape(data,(64,64,64,3)), interpolate=True, mipmap=False )
  File "/usr/lib/python2.7/dist-packages/visual/materials.py", line 70, in __init__
    self.__setattr__(key, value)
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
[display_3D_visualization_node-2] process has died [pid 5044, exit code 1, cmd /home/amitsinha/catkin_ws/src/razor_imu_9dof/nodes/display_3D_visualization.py __name:=display_3D_visualization_node __log:=/home/amitsinha/.ros/log/f3115532-6588-11e8-ae14-a44cc847fdbe/display_3D_visualization_node-2.log].
log file: /home/amitsinha/.ros/log/f3115532-6588-11e8-ae14-a44cc847fdbe/display_3D_visualization_node-2*.log

```

The visualization portion doesn't work. On looking into it further, I get the error in the python shell when I try to import the visual module:
```
$python
Python 2.7.12 (default, Dec  4 2017, 14:50:18) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import visual
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/visual/__init__.py", line 66, in <module>
    from visual.ui import display
  File "/usr/lib/python2.7/dist-packages/visual/ui.py", line 3, in <module>
    import materials
  File "/usr/lib/python2.7/dist-packages/visual/materials.py", line 152, in <module>
    tx_turb3 = raw_texture( data=reshape(data,(64,64,64,3)), interpolate=True, mipmap=False )
  File "/usr/lib/python2.7/dist-packages/visual/materials.py", line 70, in __init__
    self.__setattr__(key, value)
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
>>> 

```

Any alternate suggestions on visualization? Something like rviz or anything.

Thanks

andrestoga, 2018-07-04:
Hi,

Which version of Ubuntu/ROS are you using?

amitfishy, 2018-07-04:
I am using Ubuntu 16 with ROS Kinetic.

Thanks,
Amit

andrestoga, 2018-07-30:
I really don't know what's going on because I installed it in another computer and it works. The reason could be that ros is not loading the python library that corresponds to the python version.

amitfishy, 2018-08-02:
Actually it is not working in the Python environment itself when I try a:
```
>>> import visual
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named visual

```

Could you please tell me how you installed the python visual library? I'm not able to find documentation on it for some reason.

Thanks,
Amit

andrestoga, 2018-08-04:
I installed Anaconda that comes with VPython

https://www.anaconda.com/download/#linux

andrestoga, 2018-08-05:
Apparently, it is not possible to use VPython with ROS because it only runs on the jupyter notebook... and I don't know how to fix the error that you are having. On the other hand, the IMU orientation can be easily visualized by broadcasting its transformation and displaying it on RViz.

amitfishy, 2018-08-06:
Thanks for the help, I did try installing VPython and then trying the import but it didn't really recognize it, I guess RViz is just good enough then.

closed 2018-08-06 by amitfishy

AhmedAbdlhamid, 2019-02-17:
you can check this out , this is an updated version using visual python and ROS in anaconda.also the export file of the environment and guide to run  is included.
https://github.com/AhmedAbdlhamid/IMU_Razor9DOF