ENSTABretagneRobotics/razor_imu_9dof #40: Does this work with the new M0 Razor IMU?

Shreeyak, 2018-06-15:
Hey, the old razor imu is now a discontinued product and can't be found anywhere. It's been replaced with the new M0 Razor IMU. Does this repo work with it? Are there any plans to make it compatible with the new one? 

Also, could I please request you to mention something like this in the readme, and perhaps put an image of the board in the readme? It'd help a lot!
Thanks!

Shreeyak, 2018-06-15:
Ah, never mind. I guess I didn't notice you've mentioned M0 in the readme. My bad.

closed 2018-06-15 by Shreeyak

subodh-malgonde, 2018-07-11:
Even though you might have figured out the answer by now, I will post some additional instructions/clarifications. IMO this is important since the instructions at [razor_imu_9dof ROS Wiki](http://wiki.ros.org/razor_imu_9dof) are outdated. 

Its mentioned in the README that:
>you will need to follow the same instructions as for the default firmware on https://learn.sparkfun.com/tutorials/9dof-razor-imu-m0-hookup-guide and use an updated version of SparkFun_MPU-9250-DMP_Arduino_Library from https://github.com/lebarsfa/SparkFun_MPU-9250-DMP_Arduino_Library (an updated version of the default firmware is also available on https://github.com/lebarsfa/9DOF_Razor_IMU).

To summarize this means that, from the [Sparkfun hookup guide](https://learn.sparkfun.com/tutorials/9dof-razor-imu-m0-hookup-guide) one needs to follow the instructions from section [Installing the 9DoF Razor Arduino Core](https://learn.sparkfun.com/tutorials/9dof-razor-imu-m0-hookup-guide#installing-the-9dof-razor-arduino-core) to [Libraries and Example Firmware](https://learn.sparkfun.com/tutorials/9dof-razor-imu-m0-hookup-guide#libraries-and-example-firmware). However instead of using the [SparkFun MPU-9250 Digital Motion Processing (DMP) library](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library), one needs to use the updated version from the README. So basically the zip file of the latest release from https://github.com/lebarsfa/SparkFun_MPU-9250-DMP_Arduino_Library/releases. And instead of using the example firmware, use the firmware from this repo.