ENSTABretagneRobotics/razor_imu_9dof #41: ImportError: No module named razor_imu_9dof.cfg

andrestoga, 2018-07-04:
Hi,

I'm having the following error while running the imu_node.py:

from razor_imu_9dof.cfg import imuConfig
ImportError: No module named razor_imu_9dof.cfg

I looked in the repos and indeed there is no file named razor_imu_9dof.cfg. Only imu.cfg

My workaround was to comment out the lines of the node code where a variable of that file is used. Of course, the Dynamic Reconfigure is not available because of that.

Any help would be appreciated.

P.S. I'm using Ubuntu 16.04 with Kinetic.

Best,

andrestoga, 2018-07-04:
Hi,

I forgot to build the package with catkin_make in order to generate the files for the dynamic reconfigure.

Best,

closed 2018-07-04 by andrestoga

subodh-malgonde, 2018-07-11:
I had the same problem and my case I forgot to `source devel/setup.bash`