ENSTABretagneRobotics/razor_imu_9dof #43: Wrong coordinate system for SEN 14001?

subodh-malgonde, 2018-07-20:
I am using Razor 9DOF IMU (hardware version [SEN14001](https://www.sparkfun.com/products/14001?_ga=2.53386280.1490126341.1532006074-1776798767.1529065035&_gac=1.159316424.1529957755.Cj0KCQjwpcLZBRCnARIsAMPBgF0tmzdW0XDwz0mtrEYDW566anwqDR984fDvdBJNIVJSJlSAkKHAtBIaAiJeEALw_wcB)). The code in [imu_node.py](https://github.com/KristofRobot/razor_imu_9dof/blob/indigo-devel/nodes/imu_node.py) converts the readings from Razor AHRS firmware to ROS standards ([REP 103](http://www.ros.org/reps/rep-0103.html)). However when I test the orientations by measuring the acceleration due to gravity (9.8 m/s^2) on each axis, I get very different orientations. See these Pictures:

[image]
[image]
[image]

### This means the axis are oriented this way:

[image]

This is the left handed coordinate system, whereas ROS uses right handed coordinate system.

Is this repo supposed to work with SEN14001?

subodh-malgonde, 2018-07-20:
I observed values from the **serial monitor**, which shows data directly from Razor_AHRS, I get this coordinate system:

[image]

So it looks like  Razor_AHRS has the same coordinate system as [REP-103](http://www.ros.org/reps/rep-0103.html), which is: 

1. X axis pointing forward (towards the short edge with the connector holes)
2. Y axis pointing to the left
3. Z axis pointing up

So `imu_node.py` does not need to modify anything. Am I right @lebarsfa?

lebarsfa, 2018-07-20:
This is a bit unexpected... The conventions should be the one described here : https://github.com/Razor-AHRS/razor-9dof-ahrs/issues/57#issuecomment-378585065 ... See also https://github.com/Razor-AHRS/razor-9dof-ahrs/wiki/tutorial (around the middle of the page, the screenshot of Razor_AHRS_test.pde). With the serial monitor, do you get yaw=0 when towards the North, yaw=90 towards the East, yaw=-90 (or 270) towards the West? If it is not the case, that means that you probably use a different/modified firmware inside the Razor (also, do you use the default options?), or maybe your Razor is different...

For the accelerations, note this : https://github.com/KristofRobot/razor_imu_9dof/blob/fb23b76e4727cd294a84c453506ed6d5eeb7562e/nodes/imu_node.py#L240
This probably explains why the sign of the accelerations is the contrary.

subodh-malgonde, 2018-07-23:
> With the serial monitor, do you get yaw=0 when towards the North, yaw=90 towards the East, yaw=-90 (or 270) towards the West?

I get yaw = 90 towards east. However when I orient it to north, it does not go to 0. Also for west it does not go to -90. I think my magnetometer is faulty. See this GIF, as I rotate the Razor around the Z axis, the visualitzation does not follow the actual movement.

[image]

I used [Razor_AHRS_test.pde](https://github.com/Razor-AHRS/razor-9dof-ahrs/blob/master/Processing/Razor_AHRS_test/Razor_AHRS_test.pde) for the above sketch. For rotations around X & Y axes, it follows (this part is not in the GIF).

>If it is not the case, that means that you probably use a different/modified firmware inside the Razor (also, do you use the default options?), or maybe your Razor is different...

I have not changed anything from `Razor_AHRS.ino` apart from `HW__VERSION_CODE` (14001). Is that what you mean by default options?  

I think the firmware inside the Razor is different, maybe Sparkfun changed it.

subodh-malgonde, 2018-07-23:
**UPDATE**: It turns out that the incorrect yaw on the horizontal plane was due to non-calibrated sensors. This resulted in the IMU firmware applying incorrect drift correction for the gyroscope and magnetometer.  Source: [Troubleshooting: Unstable yaw (aka heading aka azimuth) readings / Yaw drift](https://github.com/Razor-AHRS/razor-9dof-ahrs/wiki/tutorial#troubleshooting). After calibrating the sensors, my `yaw` values are correct.

Going back to the original problem of wrong co-ordinate system, I applied these corrections to the code in `imu_node.py` to fit it:

1. https://github.com/subodh-malgonde/razor_imu_9dof/commit/bf06e0265061139f09dd1cce2b710029974f91b9
2. https://github.com/subodh-malgonde/razor_imu_9dof/commit/457f7aaff77aed552bc51cacff894cea1239a95e
3. https://github.com/subodh-malgonde/razor_imu_9dof/commit/0c0aa562a2a79ae359382786290e469ffad0e57f

closed 2018-07-23 by subodh-malgonde