ENSTABretagneRobotics/razor_imu_9dof #46: Using IMU at computer powerup

swiz23, 2018-12-13:
Hi,

I'm using the SEN 14001 imu on Ubuntu 16.04. If I plug it into a powered-on computer, it works fine. But, if I plug it into the computer while the computer is off, then turn on the computer, the device doesn't function properly. For example, it will stream out the default YPR angles to the Arduino serial monitor. However, if I try to send a command to it (llike #o0 or #p or the other commands), it completely ignores them. To get around this, I've been unplugging and replugging the imu to the already-powered-on computer. But, I feel that this should be better handled.

joekeo, 2019-09-30:
I am having the same issue. I am yet to explore into modifying the arduino sketch. I will do that later in the week and let you know if it worked.

swiz23, 2019-10-10:
Thanks! I'm no longer working on that proejct, but I'd still be interested in seeing what you come up with!

joekeo, 2019-10-29:
Hello swiz23, 
I mae a PR to fix this issue. feel free to check it out and comment on it.
https://github.com/KristofRobot/razor_imu_9dof/pull/54