ENSTABretagneRobotics/razor_imu_9dof #48: IMU Locking to a certain direction

1140452, 2019-04-02:
Hi, I have a machine with ubuntu 16.04 with kinetic installed. I'm trying to use the 9dof razor imu, version Sen 14001 and I installed multiple times, flashed the IMU with different firmwares and progams, but I end up with the same problem, which is the IMU locking it's direction to the East, when I'm visualizing it in ROS. 
However, checking the YPR in the Arduino serial, I checked and believe the value are correct. 
With that said, I don't if there's a step I'm missing or if there is a problem with ROS obtaining the values.

Also by curiosity, I calibrated the imu a few times and the gyroscope always gives 0.00 or 0.01 when in the examples it has some quite different values, is it because of a special reason?
Thx.

1140452, 2019-04-03:
I'm closing this issue because I managed to make it work, however I don't know what the solution actually was, I just started to get the values of the orientation correctly. I'll assume there was a problem somewhere with a configuration or something like that.

closed 2019-04-03 by 1140452

boheyse, 2019-04-05:
Hello, could you possibly help me with the issue I just raised in this repository?