ENSTABretagneRobotics/razor_imu_9dof #56: Create ROS2 branch

klintan, 2020-01-18:
I'm working on a ROS2 port, would be awesome if you could create a ROS2 branch for future PRs.

Thanks in advance!

mhaboali, 2020-02-22:
Hi @klintan 

I've used the ROS-bridge to get the IMU data onto the ROS2 side, it works for me but I don't know exactly what advantages will I get when I use ROS2 driver for the imu sensor. Please share with me your thoughts.

Regards, 
Mohamed

klintan, 2020-02-22:
> Hi @klintan
> 
> I've used the ROS-bridge to get the IMU data onto the ROS2 side, it works for me but I don't know exactly what advantages will I get when I use ROS2 driver for the imu sensor. Please share with me your thoughts.
> 
> Regards,
> Mohamed

In general, the last version of ROS1 is around the corner, then all efforts will be put on ROS2. That is one big reason to port everything needed to ROS2. 

Further I guess you'll get some overhead in terms of serializing and deserializing if you are using the bridge (probably not a huge problem, depending on your application).

Further just having both ROS1 and ROS2 installation, is somewhat of a hassle, launch systems etc to coordinate stuff if used for one robot is not ideal either.

The ROS2 version is done though, so feel free to test if you want to :) 
https://github.com/klintan/ros2_razor_imu

mhaboali, 2020-02-25:
> In general, the last version of ROS1 is around the corner, then all efforts will be put on ROS2. That is one big reason to port everything needed to ROS2.
>
> Further I guess you'll get some overhead [...]

Yes, I got what you mean, you're right. Thanks for the explanation!

Thanks for sharing with me the ROS2 version, I'll test it soon and I'll let you know how it will work for me :D

zoldaten, 2023-08-30:
@klintan it seems not working (ros2-rolling):
```
rock@rock-5b:~$ ros2 launch ros2_razor_imu razor-pub.launch.py
[INFO] [launch]: All log files can be found below /home/rock/.ros/log/2023-08-30-16-17-51-238535-rock-5b-35961
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - cannot import name 'get_default_launch_description' from 'launch_ros' (/opt/ros/rolling/lib/python3.10/site-packages/launch_ros/__init__.py)
 - The launch file may have a syntax error, or its format is unknown
```

even with direct start:
```
rock@rock-5b:~/ros2_ws/src/ros2_razor_imu/src/ros2_razor_imu$ python3 imu_node.py
[INFO] [1693401684.891938865] [imu_node]: Razor IMU -> Opening /dev/ttyACM0...
[ERROR] [1693401684.892600646] [imu_node]: Razor IMU not found at port /dev/ttyACM0. Did you specify the correct port in the launch file? Trying 5 more times...
[ERROR] [1693401687.896478511] [imu_node]: Razor IMU not found at port /dev/ttyACM0. Did you specify the correct port in the launch file? Trying 4 more times...
```

in minicom on port /dev/ACM0 i see dataflow:
```
#YPR=34.23,5.97,174.02
#YPR=34.25,6.28,174.29
#YPR=34.24,6.26,174.03
#YPR=34.25,6.07,174.12
#YPR=34.26,6.31,174.19
#YPR=34.24,6.14,173.93
#YPR=34.32,6.23,174.21
#YPR=34.32,6.38,174.13
```

Megaax, 2023-10-19:
> @klintan it seems not working (ros2-rolling):
> 
> first of all you have got wrong spelling: not `ros2 launch razor_imu_9dof razor-pub-and-display.launch.py` but `ros2 launch ros2_razor_imu razor-pub.launch.py`
> 
> secondly, it`s not working with launch:
> 
> ```
> rock@rock-5b:~$ ros2 launch ros2_razor_imu razor-pub.launch.py
> [INFO] [launch]: All log files can be found below /home/rock/.ros/log/2023-08-30-16-17-51-238535-rock-5b-35961
> [INFO] [launch]: Default logging verbosity is set to INFO
> [ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
>  - cannot import name 'get_default_launch_description' from 'launch_ros' (/opt/ros/rolling/lib/python3.10/site-packages/launch_ros/__init__.py)
>  - The launch file may have a syntax error, or its format is unknown
> ```
> 
> nor with direct start:
> 
> ```
> rock@rock-5b:~/ros2_ws/src/ros2_razor_imu/src/ros2_razor_imu$ python3 imu_node.py
> [INFO] [1693401684.891938865] [imu_node]: Razor IMU -> Opening /dev/ttyACM0...
> [ERROR] [1693401684.892600646] [imu_node]: Razor IMU not found at port /dev/ttyACM0. Did you specify the correct port in the launch file? Trying 5 more times...
> [ERROR] [1693401687.896478511] [imu_node]: Razor IMU not found at port /dev/ttyACM0. Did you specify the correct port in the launch file? Trying 4 more times...
> ```
> 
> in minicom on port /dev/ACM0 i see dataflow:
> 
> ```
> #YPR=34.23,5.97,174.02
> #YPR=34.25,6.28,174.29
> #YPR=34.24,6.26,174.03
> #YPR=34.25,6.07,174.12
> #YPR=34.26,6.31,174.19
> #YPR=34.24,6.14,173.93
> #YPR=34.32,6.23,174.21
> #YPR=34.32,6.38,174.13
> ```
> 
> ps. fixed only a node (not launch file): 1. `nano /home/rock/ros2_ws/src/ros2_razor_imu/src/ros2_razor_imu/imu_node.py ` changed threre to `port = self.declare_parameter('port', '/dev/ttyACM0').value` 2. fixed permission to serial: `sudo chmod 666 /dev/ttyACM0` 3. have to install:
> 
> ```
> pip3 install pyserial
> pip3 install cryptography
> pip3 install vpython
> sudo apt install libgtk-3-dev  
> pip3 install -U -f https://extras.wxpython.org/wxPython4/extras/linux/gtk3/ubuntu-20.04 wxPython --user
> pip3 install transforms3d
> ```
> 
> and it seems it`s working by direct start - python3 imu_node.py
> 
> ```
> angular_velocity:
>   x: -0.03
>   y: 0.01
>   z: -0.01
> angular_velocity_covariance:
> - 0.002
> - 0.0
> - 0.0
> - 0.0
> - 0.002
> - 0.0
> - 0.0
> - 0.0
> - 0.002
> linear_acceleration:
>   x: 1.7620156249999999
>   y: 0.388026484375
>   z: -10.868955078125
> linear_acceleration_covariance:
> - 0.04
> - 0.0
> - 0.0
> - 0.0
> - 0.04
> - 0.0
> - 0.0
> - 0.0
> - 0.04
> ---
> ```

can you please upload the imu_node.py that is working

zoldaten, 2023-10-19:
[imu_node.zip](https://github.com/ENSTABretagneRobotics/razor_imu_9dof/files/13043083/imu_node.zip)

AhmedBaza1, 2023-11-17:
I forked this repo solved the problems mentioned and made a pull request for my updates, meanwhile you can check the updates here: 
https://github.com/AhmedBaza1/ros2_razor_imu.git