ENSTABretagneRobotics/razor_imu_9dof #63: UnicodeDecodeError when running  roslaunch razor_imu_9dof razor-pub.launch

AES0007, 2022-01-26:
Good Day,

I have the Open_Log Artemis IMU DEV-16832, and followed the steps below: 

1. Update Firmware to V10.21 via GIU  - **Successful**
2. Checked to see the IUM functions (via Arduino, serial monitor, port 115200)  - **Successful** 
3. download and build sudo apt-get install ros-melodic-razor-imu-9dof   - **Successful**
4. Tried to run _roslaunch razor_imu_9dof razor-pub.launch_   - **Failed with the error below**

### Error: 
[**INFO] [1643203227.324861]: Flushing first 200 IMU entries...
Traceback (most recent call last):
  File "/opt/ros/melodic/share/razor_imu_9dof/nodes/imu_node.py", line 220, in <module>
    line = bytearray(ser.readline()).decode("utf-8")
  File "/usr/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeDecodeError: 'utf8' codec can't decode byte 0xbd in position 0: invalid start byte
[imu_node-2] process has died**

Please advise on how I can resolve. 

Regards,
Che

AES0007, 2022-01-27:
This is the data log file contents:

01/01/2000,00:41:52.44,-3.91,-126.46,-999.02,-2.35,0.57,0.40,-73.50,-198.90,-127.65,31.91,10.055,
01/01/2000,00:41:52.54,-2.93,-133.79,-997.07,-0.36,2.59,-0.11,-74.55,-200.10,-129.15,32.10,10.055,
01/01/2000,00:41:52.64,-0.49,-132.81,-1005.86,0.95,1.63,-3.92,-73.95,-199.50,-128.55,32.10,10.054,
01/01/2000,00:41:52.74,-7.81,-133.30,-988.28,-0.27,0.97,0.92,-74.85,-198.60,-128.70,32.10,10.054,
01/01/2000,00:41:52.84,2.44,-128.42,-991.70,0.63,1.97,3.62,-73.95,-198.0,-128.85,31.91,10.054,
01/01/2000,00:41:52.94,7.81,-124.51,-991.21,1.43,2.31,3.14,-73.65,-197.55,-128.55,32.20,10.053,
01/01/2000,00:41:53.04,0.98,-118.65,-997.07,-2.55,-1.50,-4.28,-73.65,-198.0,-129.60,31.91,10.053,
01/01/2000,00:41:53.14,8.30,-134.28,-981.93,0.08,2.15,-0.20,-74.55,-198.90,-130.50,32.25,10.053,
01/01/2000,00:41:53.24,10.25,-131.35,-983.89,-0.04,-1.59,-3.77,-75.45,-198.30,-131.10,32.25,10.052,
01/01/2000,00:41:53.34,14.65,-127.93,-1004.88,-0.11,4.55,-0.24,-74.40,-197.25,-129.60,32.10,10.052,
01/01/2000,00:41:53.44,16.60,-124.51,-996.09,0.02,3.63,0.82,-74.55,-198.15,-130.80,32.06,10.052,
01/01/2000,00:41:53.54,12.21,-128.42,-990.23,-0.81,0.66,1.69,-74.40,-196.50,-131.10,32.10,10.052,
01/01/2000,00:41:53.64,17.58,-116.70,-993.65,-0.79,0.80,-2.47,-75.30,-198.90,-131.55,32.01,10.051,
01/01/2000,00:41:53.74,9.28,-119.63,-996.09,2.27,1.76,0.56,-76.65,-197.10,-132.60,32.10,10.051,

sontung1010, 2024-04-17:
Hi can someone solve this because I also have the same problem

lebarsfa, 2024-04-18:
Please double-check that you correctly did the step https://github.com/ENSTABretagneRobotics/razor_imu_9dof#install-arduino-firmware since this ROS node is not made for the factory default firmware of the OpenLog Artemis.