ENSTABretagneRobotics/razor_imu_9dof #64: Potential vulnerability: topic name from a ROS parameter

tandf, 2022-12-29:
Hi,

We notice that you are using topic name from a ROS parameter at the following locations:
﻿https://github.com/ENSTABretagneRobotics/razor_imu_9dof/blob/b25b43e8244d30122bbae4f4771568d2f4f47be4/nodes/imu_node.py#L132
For security reasons detailed below, we strongly suggest avoiding the usage of strings from parameters as topic names.

Although parameters are usually set in parameter files, they can also be changed by nodes. Specifically, other nodes in the same ROS application can also change the parameters listed above before it’s used, either by accident or *intentionally* (i.e., by potential attackers). If the `razor_imu_9dof/topic` parameter is changed, the *razor_imu_9dof* node will publish to a wrong topic, which breaks the dependency of downstream nodes. Depending on the functionality of the victim robot, the victim robot may have problem localizing and navigating. Moreover, if an attacker exists, she can control and inject malicious IMU data into the victim robot, by first fooling the *razor_imu_9dof* node to publish to a wrong topic like `/imu_fake`, and then forward all messages from `/imu_fake` to `/imu` after modifying the contents. The attacker may be able to control the movement of the robot, and navigate it to arbitrary destinations. Because ROS is an OSS (open-source software) community, third-party nodes are widely used in ROS applications, usually without complete vetting of their behavior, which gives the opportunity to potentially malicious actors to inject malicious code (e.g, by submitting hypocrite commits like in other OSS systems [1]) to infiltrate the ROS applications that use it (or software supply chain attacks, one of the primary means for real-world attackers today [2]).

We understand that using parameters to set topic names brings flexibility. Still, for the purpose of security, we strongly suggest that you avoid such vulnerable programming patterns if possible. For example, to avoid the exposure of this specific vulnerability, you may consider alternatives like remapping, which is designed for configuring names when launching the nodes.

[1] Q. Wu and K. Lu, “On the feasibility of stealthily introducing vulnerabilities in open-source software via hypocrite commits,” 2021, https://linuxreviews.org/images/d/d9/OpenSourceInsecurity.pdf.
[2] Supply chain attacks are the hacker’s new favourite weapon. and the threat is getting bigger. https://www.zdnet.com/article/supply-chain-attacks-are-the-hackers-new-favourite-weapon-and-the-threat-is-getting-bigger/.

tandf, 2023-03-26:
Hi there, I wanted to follow up on this security vulnerability. Could you please let me know if there have been any updates or concerns regarding this issue? Thanks

lebarsfa, 2023-03-27:
Thank you for the report, at the moment we did not have the time to think about it but we keep that in mind!